dagoma3d/CuraByDagoma #1: Mises à jour des personnalisations faites pour 15.04.3 et discoeasy200

arthurlutz, 2017-01-19:
https://dagoma.fr/logiciels.html affiche des logiciels pour mis à jour pour la discoeasy200 et des nouvelles versions de cura. Pourtant ce dépôt n'a pas bougé depuis 2 ans. 

Je pense que ça pourrait être profitable à vos utilisateurs et à votre image de publier ces personnalisations ici (au delà du fait que la https://github.com/Ultimaker/Cura/blob/master/LICENSE vous y oblige probablement), à moins que ce ne soit publié ailleurs ?

dagomafr, 2018-01-24:
Bonjour Arthur,

Tu as tout à fait raison, cela va être fait bientôt.

Cordialement,
Orel

arthurlutz, 2018-01-24:
Très bonne nouvelle. Merci d'avance.

arthurlutz, 2020-06-19:
pouvez vous m'indiquer pourquoi ce ticket a été fermé ? est-il en "wontfix" ou résolu ?

0r31, 2020-06-19:
Bonjour Arthur,

Il est résolu. Toutes les sources sont à jour que ce soit pour CuraByDagoma ou CuraEngine.
Cela fait longtemps que c'est le cas.

dagoma3d, 2020-06-19:
Ce ticket est clos car les sources publiées ici sont maintenant à jour.

closed 2020-06-19 by dagoma3d