dagoma3d/CuraByDagoma #2: Affichage cassé sur Pop!_OS

RossinesP, 2019-12-09:
Sur Pop!_OS 19.10 sur un HP Spectre x360 (écran 4k), l'affichage est "cassé" au niveau de la zone de prévisualisation de ce qui va être imprimé.
La prévisualisation 3D s'affiche sur environ un quart de la zone, en bas à gauche. Je peux voir l'ensemble de l'image (imprimante + pièce à imprimer), c'est juste que c'est affiché en petit, en bas à gauche. Le reste de la zone est grise.
La fenêtre prend les inputs de souris correctement (c'est à dire, si je clique environ au centre de la zone d'affichage grise, ça sélectionne bien l'objet au centre de la zone réellement affichée).

RossinesP, 2019-12-09:
Voir ici : https://ibb.co/7y52Rz7

RossinesP, 2019-12-20:
C'est le mode HiDPI qui casse l'affichage, si je désactive ce mode, l'affichage fonctionne correctement.

0r31, 2020-06-19:
Bonjour,

J'arrive un peu en retard :( Je regarderai ce qui peut provoquer cela.

0r31, 2022-06-29:
Devrait normalement être fixé par le commit suivant : https://github.com/dagoma3d/CuraByDagoma/commit/84c71cb1ecf292a733fafcbd66a4ffde9d40dbe7

Et en particulier par l'utilisation de [GetContentScaleFactor()](https://docs.wxpython.org/wx.Window.html#wx.Window.GetContentScaleFactor).

Testé sur MacBook 4k

closed 2022-07-01 by 0r31