dagoma3d/CuraByDagoma #3: Support Ubuntu 20.40

dadall, 2020-10-06:
Bonjour,

Avez-vous l’intention de supporter Ubuntu 20.04 pour qu'on évite de passer par[ cette magouille](https://support.dagoma.fr/support/solutions/articles/36000227539-curabydagoma-ne-fonctionne-plus-sous-ubuntu-20-04) ?

Cordialement,

0r31, 2020-10-06:
Bonsoir @dadall ,

Ce n'est pas vraiment une magouille. Certains paquets python2 ont été retirés des sources list d'ubuntu au passage de la 20.04 (certainement lié à l'arrêt du support de python2 depuis le début de l'année).

Je travaille sporadiquement sur le portage python3 et si possible la mise à disposition d'une AppImage.
Voir la branche dagoma3d/CuraByDagoma/tree/migration_python3

Cordialement,
Orel

dadall, 2020-10-06:
Bonsoir,

Je regrette d'avoir raté la branche, ça m'aurait évité de « râler » bêtement !

M'en vais tester tout ça.

Bonne soirée,

closed 2020-10-06 by dadall

0r31, 2020-10-06:
Re bonsoir @dadall ,

Attention, le travail sur le sujet n'est pas terminé. Il est possible que tu rencontres quelques problèmes.
Je t'invite, si tu le souhaites, à venir sur le [discord de Dagoma](https://discord.gg/CVMSRX) pour en discuter.

Cordialement,
Orel