dagoma3d/CuraByDagoma #4: Bug "layers" curabydagoma avec python3

vledoze, 2020-11-05:
Bonjour, 

J'utilise Ubuntu 20.04 et j'ai tenté la version avec python3. Lorsque j'essaye de voir les "layers" de ma pièce, cura plante avec le message d'erreur suivant: 

```
Traceback (most recent call last):
  File "/usr/share/curabydago/Cura/util/sliceEngine.py", line 430, in _watchProcess
    self._process = self._runEngineProcess(commandList)
  File "/usr/share/curabydago/Cura/util/sliceEngine.py", line 699, in _runEngineProcess
    return subprocess.Popen(cmdList, stdin=subprocess.PIPE, stdout=subprocess.PIPE, stderr=subprocess.PIPE, encoding='utf-8', **kwargs)
  File "/usr/lib/python3.8/subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.8/subprocess.py", line 1702, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: '/usr/share/curabydago/CuraEngine'
Traceback (most recent call last):
  File "/usr/share/curabydago/Cura/gui/util/openglGui.py", line 227, in _OnGuiPaint
    self.OnPaint(e)
  File "/usr/share/curabydago/Cura/gui/sceneView.py", line 1206, in OnPaint
    self._engineResultView.OnDraw()
  File "/usr/share/curabydago/Cura/gui/util/engineResultView.py", line 85, in OnDraw
    if len(result._polygons) > layerNr-1 and 'inset0' in result._polygons[layerNr-1] and len(result._polygons[layerNr-1]['inset0']) > 0 and len(result._polygons[layerNr-1]['inset0'][0]) > 0:
AttributeError: 'NoneType' object has no attribute '_polygons' 
```
Bonne journée

0r31, 2020-11-05:
Bonsoir @vledoze ,

Ce sera corrigé dans la prochaine version qui sortira officiellement ce soir ou demain.

dagoma3d, 2020-11-05:
Bonsoir @vledoze ,

Tu trouveras une pre-release ici : https://github.com/dagoma3d/CuraByDagoma/releases/download/2.1.7-beta-python3/CuraByDagoma_amd64.deb

vledoze, 2020-11-06:
Super ! 
Merci :)

closed 2020-11-06 by vledoze