dagoma3d/CuraByDagoma #5: Mac OS notarize for Big Sur

bourvill, 2020-11-23:
Hello
Cura by Dagoma can't be launch since Big Sur release.

Developer isn't identified by apple via notarize service

0r31, 2020-11-23:
Hi,

We never signed the application.
Before Big Sur, the same behavior happened. Maybe the process to open an unidentified application has changed introducing this new release.
Please check instructions from Apple here : https://support.apple.com/fr-fr/guide/mac-help/mh40616/mac

On our side, we will check if signing the app can be done.

Regards,
Orel

Alutulu, 2022-06-30:
To create an Apple ID : https://appleid.apple.com/account

Alutulu, 2022-06-30:
Seems to be fixed

0r31, 2022-07-01:
For MacOS, we have to disable Gatekeeper for this application. After installing it (default location : /Applications), please type the following command in a terminal :
xattr -cr /Applications/CuraByDagoma.app
I hope we will be able to sign the application asap...

VincentMasselis, 2022-08-21:
On MacOS Monterey, the system displays a popup error about the dmg integrity. It took me several minutes before finding why I've this exception.

IMO, notarize the app will be great for the user experience.