dagoma3d/CuraByDagoma #7: Examples trigger "struct.error: unpack requires a buffer of 50 bytes"

lissyx, 2020-12-16:
Hello,

Recently bought a Disco Ultimate, and following the doc I'm trying to print the holders examples provided at https://start.dagoma3d.com/printer/du/notice-utilisation:
```
$ ll *.STL
-rw-rw-r-- 1 alex alex 9,1M mai   25  2018 Base-spoolholder.STL
-rw-rw-r-- 1 alex alex  22M mai   25  2018 Top-spoolholder.STL
```

When loading both in CuraByDagoma (latest version) this fails and one can see on the console:
```
$ curabydago 
Load preferences from /home/alex/.curaByDagoma/2.1.7/preferences.ini
17:05:06: Debug: Adding duplicate image handler for 'Windows bitmap file'
Listening for engine communications on 49674
Traceback (most recent call last):
  File "/usr/share/curabydago/Cura/gui/sceneView.py", line 739, in loadScene
    objList = meshLoader.loadMeshes(filename)
  File "/usr/share/curabydago/Cura/util/meshLoader.py", line 32, in loadMeshes
    return stl.loadScene(filename)
  File "/usr/share/curabydago/Cura/util/meshLoaders/stl.py", line 61, in loadScene
    _loadBinary(m, f)
  File "/usr/share/curabydago/Cura/util/meshLoaders/stl.py", line 47, in _loadBinary
    data = struct.unpack("<ffffffffffffH", f.read(50))
struct.error: unpack requires a buffer of 50 bytes
```

Env:
 - Ubuntu 20.10
 - https://github.com/dagoma3d/CuraByDagoma/releases/download/2.1.7-python3/CuraByDagoma_amd64.deb

Some other (simpler) examples works well. Is this a bug in Cura/CuraByDagoma?

lissyx, 2020-12-16:
FTR, Windows x64 py2 opens theffile successfully.

0r31, 2020-12-17:
Hi @lissyx ,

Thanks for your feedback, will take a look and come back to you.

Regards,
Orel

lissyx, 2020-12-17:
It seems to be because of the end of the file specifically:
```
>>> f_data = b'\n      endloop\r\n   endfacet\r\nendsolid'
>>> struct.unpack("<ffffffffffffH", f_data)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
struct.error: unpack requires a buffer of 50 bytes
>>> 
```

lissyx, 2020-12-17:
Ok, it looks like it's going through binary mode for an ascii file.

lissyx, 2020-12-17:
Looks like it's trivially some py2/3 behavior?
```
$ python2
Python 2.7.18 (default, Apr 20 2020, 20:30:41) 
[GCC 9.3.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> f = open("/home/alexandre/Téléchargements/support-bobine/Base-spoolholder.STL", "rb")
>>> format = f.read(5)
>>> format
'solid'
>>> format.decode()
u'solid'
>>> format
'solid'
>>> format.decode() == "solid"
True
>>> 
$ python3
Python 3.9.1 (default, Dec  8 2020, 07:51:42) 
[GCC 10.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> f = open("/home/alexandre/Téléchargements/support-bobine/Base-spoolholder.STL", "rb")
>>> format = f.read(5)
>>> format
b'solid'
>>> format.decode()
'solid'
>>> format.decode() == "solid"
True
>>> 
```

0r31, 2020-12-17:
Hi,

Thanks for digging into this.
I did the migration but excepting some bug related to this topic (binary read by default in python3 not in python2).
Some pieces of code have already been fixed,... not this one :(

I am on another topic right now, but i will provide you a patch (or you could take the commit diff when done) before the end of the day.

Regards,
Orel

lissyx, 2020-12-17:
> Hi,
>
> Thanks for digging into this.
> I did the migration but excepting some bug related to this topic (binary read by default in python3 not in python2).
> Some pieces of code have already been [...]

Thanks, I meant no pressure, I was looking forward to fixing that.

lissyx, 2020-12-17:
@0r31 Here is a PR #9, it works good enough for me:
 - loads the dagoma ultimate background STL
 - loads both holder examples that would fail before
 - loads all the STL other files I could find locally

commit 27f0811a (2020-12-18): Fix #7: Proper detection of ASCII / Binary STL formats
closed 2020-12-18 by 0r31