dagoma3d/CuraByDagoma #11: Unable to save to filesystem when a ttyACM device is plugged

lissyx, 2021-01-05:
Trying to use CuraByDagoma, system thinks it has a 3D printer connected and complains it cannot print to it.

STR:
 1. Arduino project connected over USB (presented as `/dev/ttyACM0`)
 2. Load a STL
 3. Click "Prepare for printing"

Expected:
 I can print.

Actual:
 I cannot print, I get a modal dialog telling me it cannot print.

I suspect that CuraByDagoma tries to detect USB-connected printer, and that my Arduino project connected is picked up as a false positive. However, there seems to be no way to tell CuraByDagoma to bypass trying to print directly and just get me to save on the filesystem.

0r31, 2021-01-06:
Hi @lissyx ,

Indeed, there is single button to handle different behaviors (save to sd, save to filesystem, print over usb).
I will check if it is possible to get rid of false positive when any Arduino board is plugged to the computer. If not, i guess separating behaviors in two different buttons could be fair enough.

In the meantime, you can disable USB connection usage by setting _connectionGroup_ to **None** [here](https://github.com/dagoma3d/CuraByDagoma/blob/52017f2f0ed0a6a86b4b35380620367a141e8dcc/Cura/gui/sceneView.py#L294).

Regards,
Orel

lissyx, 2021-01-06:
> I will check if it is possible to get rid of false positive when any Arduino board is plugged to the computer. If not, i guess separating behaviors in two different buttons could be fair enough.

I have not checked how the printer communicates over serial, but if there's no way to distinguish, maybe its easier to augment the error message dialog with a button "Let me save of filesystem instead".

0r31, 2021-01-06:
Indeed, could be a better option.

lissyx, 2021-01-11:
@0r31 It's also blocking me from printing from my laptop, where there's a `ttyACM0` from my LTE modem.

lissyx, 2021-01-12:
FTR, reading the code, if you have the sdcard plugged in, it will first save to it as `dagoma0.g`, so it's not such big of a deal.

0r31, 2021-01-12:
Hi @lissyx ,

Exactly. Your comment completes my previous message. Ordered choices to handle gcode :
1) save to sd card if any plugged
2) USB print if any available connection
3) otherwise save to the local filesystem

For now, I don't really have time to implement your suggestion but i will when possible.

Regards,
Orel

0r31, 2021-02-16:
Hi @lissyx,

a54240b adds a button to allow saving gcode from the print from usb window.

Regards,

_Note : the python3 migration breaks the ability to print from usb... Have to work on that._

lissyx, 2021-02-16:
> Hi @lissyx,
> 
> [a54240b](https://github.com/dagoma3d/CuraByDagoma/commit/a54240b29bad89f43f65660581ee7af0121e0064) adds a button to allow saving gcode from the print from usb window.

I can confirm it works, is it safe to close the issue?

> 
> Regards,
> 
> _Note : the python3 migration breaks the ability to print from usb... Have to work on that._

Fortunately, I don't care about that feature :)

0r31, 2021-02-16:
You can indeed close the issue if your are happy with the provided fix.

lissyx, 2021-02-16:
Perfect, thanks!

closed 2021-02-16 by lissyx