dagoma3d/CuraByDagoma #16: package.sh for debian/64 is broken

lissyx, 2021-02-16:


lissyx, 2021-02-16:
@0r31 I have a simple fix on #17, and I could successfully build a deb amd64 from `migration_python3` inside a docker image, and so far installed on my ubuntu 20.10 it works well. Obviously can't speak for other platforms.

0r31, 2021-02-16:
Hi @lissyx ,

Thanks for your feedback. I build on ubuntu 20.04 and i didn't see this error. To build a debian 64 package, the command is:
> ./package.sh debian 64 1

lissyx, 2021-02-16:
> Hi @lissyx ,
> 
> Thanks for your feedback. I build on ubuntu 20.04 and i didn't see this error. To build a debian 64 package, the command is:
> `./package.sh debian 64 1`

Well, the `help` provided when you just run `./package.sh` does not advise this last `1`

0r31, 2021-02-16:
You are right, my bad.
Will add a default value if no third argument is provided so that the engine is by default compiled.

lissyx, 2021-02-16:
(to be fair, I have not tracked down where `BUILD_ENGINE` is actually sources, so it sounds like with your fix it would make my PR useless)

0r31, 2021-02-16:
No pb.

Done in last commit 3721e74.
Will reject your PR and explain why.

lissyx, 2021-02-16:
Thanks, that was quick!

closed 2021-02-16 by lissyx