dagoma3d/CuraByDagoma #18: Clipso battery cover file cannot be opened

lissyx, 2021-09-28:
STR:
 - CuraByDago from `migration_python3` branch
 - Ubuntu 21.04
 - Download https://www.happy3d.fr/fr/cuisine/cache-pile-seb-clipso-control/
 - Try to open in CuraByDagoma

Fails to load:
```
Sep 28 10:32:07 portable-alex curabydago.desktop[2254265]: Traceback (most recent call last):
Sep 28 10:32:07 portable-alex curabydago.desktop[2254265]:   File "/usr/share/curabydago/Cura/gui/util/openglGui.py", line 184, in _OnGuiMouseDown
Sep 28 10:32:07 portable-alex curabydago.desktop[2254265]:     if self._container.OnMouseDown(e.GetX(), e.GetY(), e.GetButton()):
Sep 28 10:32:07 portable-alex curabydago.desktop[2254265]:   File "/usr/share/curabydago/Cura/gui/util/openglGui.py", line 99, in OnMouseDown
Sep 28 10:32:07 portable-alex curabydago.desktop[2254265]:     if ctrl.OnMouseDown(x, y, button):
Sep 28 10:32:07 portable-alex curabydago.desktop[2254265]:   File "/usr/share/curabydago/Cura/gui/util/openglGui.py", line 570, in OnMouseDown
Sep 28 10:32:07 portable-alex curabydago.desktop[2254265]:     self._callback(button)
Sep 28 10:32:07 portable-alex curabydago.desktop[2254265]:   File "/usr/share/curabydago/Cura/gui/sceneView.py", line 271, in showLoadModel
Sep 28 10:32:07 portable-alex curabydago.desktop[2254265]:     profile.putPreference('lastFile', filenames[0])
Sep 28 10:32:07 portable-alex curabydago.desktop[2254265]:   File "/usr/share/curabydago/Cura/util/profile.py", line 816, in putPreference
Sep 28 10:32:07 portable-alex curabydago.desktop[2254265]:     savePreferences(getPreferencePath())
Sep 28 10:32:07 portable-alex curabydago.desktop[2254265]:   File "/usr/share/curabydago/Cura/util/profile.py", line 787, in savePreferences
Sep 28 10:32:07 portable-alex curabydago.desktop[2254265]:     parser.set('preference', set.getName(), set.getValue())
Sep 28 10:32:07 portable-alex curabydago.desktop[2254265]:   File "/usr/lib/python3.9/configparser.py", line 1201, in set
Sep 28 10:32:07 portable-alex curabydago.desktop[2254265]:     super().set(section, option, value)
Sep 28 10:32:07 portable-alex curabydago.desktop[2254265]:   File "/usr/lib/python3.9/configparser.py", line 894, in set
Sep 28 10:32:07 portable-alex curabydago.desktop[2254265]:     value = self._interpolation.before_set(self, section, option,
Sep 28 10:32:07 portable-alex curabydago.desktop[2254265]:   File "/usr/lib/python3.9/configparser.py", line 402, in before_set
Sep 28 10:32:07 portable-alex curabydago.desktop[2254265]:     raise ValueError("invalid interpolation syntax in %r at "
Sep 28 10:32:07 portable-alex curabydago.desktop[2254265]: ValueError: invalid interpolation syntax in '/run/user/1000/gvfs/smb-share:server=nas.local,share=documents/Impression3D/happy3D-cache-pile-seb-clipso-control/seb%20clipso%20control%20cover.stl' at position 117
```

lissyx, 2021-09-28:
I've opened the file in FreeCAD, re-exported and it worked.

0r31, 2022-06-29:
The issue is linked with the filename containing "%" characters : seb%20clipso%20control%20cover.stl.

To reproduce : rename any stl file you have with a string containing "%" -> crash