dagoma3d/CuraByDagoma #20: Build for Windows

0r31, 2022-05-20:
We have to update all tools and processes to build a valid installer from python3 sources.

0r31, 2022-05-24:
First step : Update NSIS from 2.x to 3.x.

0r31, 2022-05-24:
Steps to initialize a ready to use virtual env : 
- Install python 3.9
- Create a virtual env : `python -m venv myvenv` and activate it.
- Install common dependencies : `pip install numpy pyserial`
- Get / install specific wheel for PyOpenGL (PyOpenGL‑3.1.6‑cp39‑cp39‑win_amd64.whl from https://www.lfd.uci.edu/~gohlke/pythonlibs/#pyopengl)
- Get  / install specific wheel for wxPython (https://files.pythonhosted.org/packages/6f/d3/155ce684967bba904a9d29c346430a491643d60a23f713fcf2fa981b7855/wxPython-4.1.1-cp39-cp39-win_amd64.whl)

0r31, 2022-05-24:
Next step will be to modify the **windows** section in [package.sh](https://github.com/dagoma3d/CuraByDagoma/blob/migration_python3/package.sh) in order to leverage the virtual environment we previously created.

closed 2022-06-03 by 0r31

BlessingCR, 2023-08-15:
The website down.  https://www.lfd.uci.edu/~gohlke/pythonlibs/#pyopengl

0r31, 2023-08-26:
[image]