dagoma3d/CuraByDagoma #33: Complety broke on last fedora (Fedora 36)

tr1p0p, 2022-05-27:
Unlike what the readme say, the linux archive are absolutly not ready to launch at all.

launching cura.py show a lots of errors of missing modules, and when you add this modules with pip, it finally show this :

`NotImplementedError: Platform does not define a GLUT font retrieval function`

It seem it's not even adress anywhere because linux don't have requirements, build instructions and all. Don't know if it's wayland fault or the fact this project use old python2, i don't really know.

I just know normal cura work and dagoma should just give the sigma profiles, because linux support is really bad it's seem.

0r31, 2022-05-27:
Hi @tr1p0p ,

For Ultimaker Cura, we are working on profiles for Sigma. After providing them, we are dependent of the Ultimaker release process so we can't tell when it will be integrated.

About CuraByDagoma, the linux archive is done from **migration_python3** branch, so it runs against python3.
Only windows version is still based on python2.

What you can do from the extracted archive is to create a python virtual env, install dependencies in it (check [requirements.txt](https://github.com/dagoma3d/CuraByDagoma/blob/migration_python3/requirements.txt and the wxPython wheel matching your system).
Some details are available here : https://github.com/dagoma3d/CuraByDagoma/tree/migration_python3#virtual-environment

I will help you as i can. I only tested on xubuntu from 16.04 to 20.04 (Xfce and Gnome).

0r31, 2022-06-01:
Hi @tr1p0p ,

As stated on the [pyopengl repo](https://github.com/mcfletch/pyopengl/issues/41), could you try explicitly define the PYOPENGL_PLATFORM then run the program from terminal?

`$ PYOPENGL_PLATFORM="egl" python3 myprogram.py`