dagoma3d/CuraByDagoma #73: Add a surface under the printer

Alutulu, 2022-06-29:
The printer is in the void for the moment : 
[image]
It would be more enjoyable if the printer was put on a surface, like the workspace in solidworks for example (with shadows) : 
[image]
Or adding a nice grid on the printer surface like ultimaker cura : 
[image]

0r31, 2022-07-01:
La grille dans ultimaker cura correspond au pavage dans curabydago.
La solution serait donc d'ajouter un "sol" (ie. un plan infini).

Alutulu, 2022-07-12:
From Manu : add a fog from a certain distance from the camera

Alutulu, 2022-07-12:
Example : 
[image]