dagoma3d/CuraByDagoma #125: Long term feasability/ lack of effectiveness for main objective

Th3Sign4l, 2022-07-30:
This application does not scale well. With the shear amount of files available, putting hash data inside a 'snow.png' file is a terrible practice when attempting to stop "weapon files" from being printed. It's also just not feasible at scale for a few reasons:

You have predefined hashes - approximately 1018 different files at the moment correct? How many files exist on the internet? If I want to defeat the program and print weapons, I'd simply have to append a character or some garbage to the file that is blocked to circumvent your "protections".

The database of files has to be updated manually - do you allow others to update the 'snow.png' file? Could there not be a list of community driven lists utilized to block files against?

The process of taking in new data manually - you're telling me you wait around for people to send STL's to you? If you really cared about getting many samples, you would do research yourself on where to get STL's, you would set up threat intelligence feeds on social media, constantly pull new files from updated sources, etc.

Consider implementing some type of learning model that observes the design of the rendered STL's. This may be computationally intensive, but you're already trying to create and maintain a project that does next to nothing to solve an issue.

I guarantee you, that if you gave me one of your printers and slicer software setups, I would 100% every time be able to print whatever I want, firearm or not.

In conclusion, 
Can't stop the signal.

0r31, 2022-07-30:
You came to the discord channel to insult people and now you come here to show off your knowledge (and you created a GitHub account just for this...).

You make a mistake on the goal of what you "discovered". Was part of a campaign, an old one, in partnership with a communication agency.
We never tried to be exhaustive for many reasons :
- The goal was to warn people
- It is not possible
- If it could be, including a check file inside the app is definitely not the right way.

And to be honest, we just use .... the same way to "hide" data that was used to distribute the famous "Liberator" in free speech communities to get around censorship.

At this point, you can do what you want : if you absolutely want to use this app, the source code is open, if not, use another...

Will be my unique answer to you.

Have a nice day.

dxgldotorg, 2022-07-31:
I think what happened was someone on Reddit stumbled upon this and didn't even look at dates before making a big deal about it.

closed 2022-08-01 by 0r31