dagoma3d/CuraByDagoma #133: np.int deprecation breaks on ubuntu 23.04

lissyx, 2023-05-03:
With latest deb package available:
```
$ curabydago
Load preferences from /home/alex/.curaByDagoma/2.2.2/preferences.ini
11:21:38: Debug: Adding duplicate image handler for 'Windows bitmap file'
11:21:38: Debug: Adding duplicate animation handler for '1' type
11:21:38: Debug: Adding duplicate animation handler for '2' type
Listening for engine communications on 49674
Traceback (most recent call last):
  File "/usr/share/curabydago/Cura/gui/util/openglGui.py", line 227, in _OnGuiPaint
    self.OnPaint(e)
  File "/usr/share/curabydago/Cura/gui/sceneView.py", line 1345, in OnPaint
    self._drawMachine()
  File "/usr/share/curabydago/Cura/gui/sceneView.py", line 1449, in _drawMachine
    meshes = meshLoader.loadMeshes(resources.getPathForMesh(machine.lower() + '.stl'))
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/curabydago/Cura/util/meshLoader.py", line 32, in loadMeshes
    return stl.loadScene(filename)
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/curabydago/Cura/util/meshLoaders/stl.py", line 64, in loadScene
    obj._postProcessAfterLoad()
  File "/usr/share/curabydago/Cura/util/printableObject.py", line 78, in _postProcessAfterLoad
    self.processMatrix()
  File "/usr/share/curabydago/Cura/util/printableObject.py", line 97, in processMatrix
    hull = numpy.zeros((0, 2), numpy.int)
                               ^^^^^^^^^
  File "/usr/lib/python3/dist-packages/numpy/__init__.py", line 305, in __getattr__
    raise AttributeError(__former_attrs__[attr])
AttributeError: module 'numpy' has no attribute 'int'.
`np.int` was a deprecated alias for the builtin `int`. To avoid this error in existing code, use `int` by itself. Doing this will not modify any behavior and is safe. When replacing `np.int`, you may wish to use e.g. `np.int64` or `np.int32` to specify the precision. If you wish to review your current use, check the release note link for additional information.
The aliases was originally deprecated in NumPy 1.20; for more details and guidance see the original release note at:
    https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations

lissyx, 2023-05-03:
Quick workaround, replace `numpy.int` with `int` as suggestes

closed 2023-05-03 by 0r31

0r31, 2023-05-04:
Before the end of the month.

lissyx, 2023-07-18:
Looks like it works well now, thanks ! :)

commit ca224a6d (2023-05-03): Fix #133