dagoma3d/CuraByDagoma #136: Double extruders not used when setting a height for color change

lissyx, 2023-10-28:
Filing as a bug but I'm not 100% sure it's one.

STR:
 - Start CuraByDagoma
 - Configure for a double-extrusion model (Disco Ultimate)
 - Load a STL file (from https://www.thingiverse.com/thing:4408535, not sure if it's file specific)
 - Select white PLA in extruder 1, black in extruder 2
 - Set start with extruder one
 - Set 33% infill, standard quality, support everywhere, add a raft (maybe all those are not required, but I wanted to be safe on this print)
 - Set to start with extruder 1
 - Use sensor
 - Click "Color change"
 - You get presented the "Layer view" with height set to the maximum of the object (59mm here)
 - Lower to 23mm (with the tracks from the example, the goal is to have white PLA for the crossties of the railtrack and black PLA for the rail itself)
 - Click "Prepare for printing", save on SD

Expected:
 At the configured height of 23mm, we should have switched from white PLA to black PLA

Actual:
 All done in white

It was my understanding you could set "Color change" with the double extrusion to automatically switch from extruder one to extruder two without having to pause, remove the filament from extruder 1 and insert a new one into extruder 1.

I know there's a way to assemble pieces for double extrusions, and according to the color used in the preview, it seems to work (tested with two different objects, used "assemble for extrusion"). In my case, even after setting color change height of 23mm, the preview was still all white.

It's possible I'm mistaken and this was never possible, but I'm pretty sure this is something I did in the very first days of owning this printer.

Please find below the preview when clicking "Color change" and setting height to 23mm on this object:
[image]

lissyx, 2023-10-30:
@0r31 By any chance do you know if I'm mistaken or did it really regress ? Or is it a feature that should be added ?

0r31, 2023-10-30:
@lissyx : i don't remember it is something we implemented before (i imagine what have to be done in sources and it doesn't trigger any light in my head). Btw, it could be a feature to add.

What has to be done :

lissyx, 2023-10-30:
> (quik and dirty way to do this is to replace the pause script by the extruder switch sequence directly in the gcode. I guess the second extruder position must be set to the next E position read)

you mean in the generated gcode of current versions ? i have no idea how complicated this is :[

0r31, 2023-10-30:
> > (quik and dirty way to do this is to replace the pause script by the extruder switch sequence directly in the gcode. I guess the second extruder position must be set to the next E position read)
> 
> you mean in the generated gcode of current versions ? i have no idea how complicated this is :[

Yep. If you want, you can send me a gcode and i will show you what I think must be done. If i am right, it is not complicated.

lissyx, 2023-10-30:
like this gcode, I've added a color switch and a pause at layer 28
[test_couleur_auto.zip](https://github.com/dagoma3d/CuraByDagoma/files/13208416/test_couleur_auto.zip)

0r31, 2023-10-30:
Thanks,

Here is how i will proceed (in waiting the soft doing it for you ;)) :
- According to the gcode comments, it is generated for a disco ultimate - 2 extruders
- The only extruder switch command it contains is `T0`, so the print is done with the first extruder
- I check if a pause at layer command is set in searching for `M600`. I see one at beginning of layer 27 (index 0)
- Now i go to the xml definition file of a disco ultimate - 2 extruders and search for the extruder switch sequences from extruder 0 to extruder 1 : https://github.com/dagoma3d/CuraByDagoma/blob/801ac6de764eace1dca8e16d909bdf9dbee40982/resources/xml/discoeasy200_dual.xml#L704

0r31, 2023-10-30:
Sorry, i ctrl+clicked ...
Now we can try to simplify the whole stuff as we don't need stuff related to prime tower. What we need is to cleanly free the nozzle from T0, then engage filament from T1.

0r31, 2023-10-30:
Here is what i have in mind :
```
G92 E0 ;Set current extruder position to 0
G1 E-60 F3000 ;Free the nozzle from filament of extruder 0
T1 ;Switch extruder
G92 E0 ;Set current extruder position to 0
G1 E60 F3000 ;Engage the filement of extruder 1 in the nozzle
G92 E1750.80156 ;Set the extruder position to the last known position
```

Here it is :
[test_couleur_auto.zip](https://github.com/dagoma3d/CuraByDagoma/files/13208651/test_couleur_auto.zip)

It is live thinking, no test.

lissyx, 2023-10-31:
Thanks! I'll give it a test. Do you know where in the code this is handled? Python side or LegacyCura? If it's the former I might try to attempt something, but if it's in LegacyCura I will likely wait (last time I tried to build the dependencies, it was just failing because my system was too new and I was lacking of time to investigate how to fix).

0r31, 2023-10-31:
It is in our codebase.
As fas as i migrate from python 2.7 from python3.x, and depending on your system, it could be possible to create a python3 virtual env then install needed dependencies in it then just run cura.py.
That's how i work. To avoid building dependencies and any errors, i try to deal with the python 3 version to be sure i have access to a wheel for thetwo main dependencies, wxPython and PyOpenGL.

lissyx, 2023-10-31:
> It is in our codebase. As fas as i migrate from python 2.7 from python3.x, and depending on your system, it could be possible to create a python3 virtual env then install needed dependencies in it [...]

`in our codebase` as in the python code of CuraByDagoma right?

As for the deps, is this section relevant? https://github.com/dagoma3d/CuraByDagoma#linux If so, then i guess I could have a look at hacking that. Welcoming any entry point in the python code though :p

lissyx, 2023-10-31:
@0r31 FTR, I've started the test print ~2h ago and from my camera what I can see is looking nice:
[image]

0r31, 2023-10-31:
> `in our codebase` as in the python code of CuraByDagoma right?

Yep, the code in this repo, master branch.

> As for the deps, is this section relevant? https://github.com/dagoma3d/CuraByDagoma#linux

Yes it is but it can be modified according to your python3 version.

> If so, then i guess I could have a look at hacking that. Welcoming any entry point in the python code though :p

[image]

0r31, 2023-10-31:
BTW, my virtual env is a python 3.10.12 one.

```console
(venv) user@hostname:~$ python --version
Python 3.10.12
```

0r31, 2023-10-31:
If you don't care about pause, you can simply replace the pause command by the switch sequence here :
https://github.com/dagoma3d/CuraByDagoma/blob/801ac6de764eace1dca8e16d909bdf9dbee40982/plugins/pauseAtZ-ByDagoma.py#L100

If you are on linux, you can directly access CuraByDagoma installation files at `/use/share/curabydago` and so you can hack here without having to change sources, package,...

lissyx, 2023-10-31:
> If you don't care about pause, you can simply replace the pause command by the switch sequence here :
> 
> https://github.com/dagoma3d/CuraByDagoma/blob/801ac6de764eace1dca8e16d909bdf9dbee40982/plugins/pauseAtZ-ByDagoma.py#L100
> 
> If you are on linux, you can directly access CuraByDagoma installation files at `/use/share/curabydago` and so you can hack here without having to change sources, then package, then install,...

nice, maybe we could just add a new "change extruder" next to "pause" to handle this ? so people can opt in.

One could imagine changing from extruder 1 to 2 then 1 again etc.

0r31, 2023-10-31:
Yep that's what i had in mind yesterday. Pretty simple, no big ui change.

lissyx, 2023-10-31:
I dont know well the risks for `wxPython` but on https://extras.wxpython.org/wxPython4/extras/linux/gtk3/ubuntu-20.04/ or https://extras.wxpython.org/wxPython4/extras/linux/gtk3/ubuntu-22.04/ (there's no 23.10) I can only see 4.2.1 release for Python 3.11, is it safe ?

0r31, 2023-10-31:
Mine is https://extras.wxpython.org/wxPython4/extras/linux/gtk3/ubuntu-20.04/wxPython-4.2.0-cp310-cp310-linux_x86_64.whl

In the meantime, I upgraded my system to 22.04. I don't see any issues for now.

lissyx, 2023-10-31:
So on 23.10 there's no more `libtiff.so.5` which the latest `wxPython4` wheel for ubuntu links against :[

0r31, 2023-10-31:
Argh :(

0r31, 2023-10-31:
symlink from `libtiff.so.x` to `libtiff.so.5`  ...?

lissyx, 2023-10-31:
I took the one from 22.04 LTS and extracted, overloading `LD_LIBRARY_PATH` and now we have a dependency against `OpenGL` python? that was not in the requirements file :(

lissyx, 2023-10-31:
Making no change, just running `python3 -m Cura.cura` from the venv ends up with this stack when I click `Color change`:
```
Traceback (most recent call last):
  File "/home/alex/codaz/CuraByDagoma/curabydagoma-git/Cura/gui/util/openglGui.py", line 227, in _OnGuiPaint
    self.OnPaint(e)
  File "/home/alex/codaz/CuraByDagoma/curabydagoma-git/Cura/gui/sceneView.py", line 1249, in OnPaint
    self._engineResultView.OnDraw()
  File "/home/alex/codaz/CuraByDagoma/curabydagoma-git/Cura/gui/util/engineResultView.py", line 85, in OnDraw
    if len(result._polygons) > layerNr-1 and 'inset0' in result._polygons[layerNr-1] and len(result._polygons[layerNr-1]['inset0']) > 0 and len(result._polygons[layerNr-1]['inset0'][0]) > 0:
           ^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute '_polygons'
^C
```

Could it be because of the wrong `wxPython4` ?

lissyx, 2023-10-31:
There are also `CuraEngine` errors, and I can see the doc covers it but `latest` release does not provide the binary: https://github.com/dagoma3d/CuraEngine/releases/tag/dagoma-2.1.9

0r31, 2023-10-31:
Ok understood, you have to have the CuraEngine binary... The "result" comes from it.
Make a symlink on the repo root root folder from the one you find in `usr/share/curabydago`

lissyx, 2023-10-31:
Thanks, it looks like I'm settled and I can hack something, maybe tonight.

lissyx, 2023-10-31:
In your example `E1750.80156` was from the lines before, I guess I need to fetch it somehow in `plugins/pauseAtZ-ByDagoma.py`.

lissyx, 2023-10-31:
(or even maybe introduce a `plugins/extruderSwitchAtZ-ByDagoma.py` ?

0r31, 2023-10-31:
Yes it comes from the last move line (G1).

Maybe a brand new plugin is the way to go. And actually, i'm not satisfied of the way the pauseAtZ is implemented :)

lissyx, 2023-10-31:
I'm getting road blocks on dumb things related to buttons :(

lissyx, 2023-10-31:
With https://github.com/dagoma3d/CuraByDagoma/pull/137
[image]

lissyx, 2023-10-31:
> Yes it comes from the last move line (G1).
> 
> Maybe a brand new plugin is the way to go. And actually, i'm not satisfied of the way the pauseAtZ is implemented :)

Ok, I created a new plugin (out of the previous sorry). I have it plugged and generating the example you shared earlier. PR draft is uptodate at https://github.com/dagoma3d/CuraByDagoma/pull/137

All it misses now is:
 - [ ] Getting correct last `G1` reference
 - [ ] Use it in the generated code
 - [ ] Dont hardcode switching between extruders multiple time

lissyx, 2023-10-31:
@0r31 ok, it's pushed and I tested on my example setting Extruder X as first one, I can see a correct sequence of switching between X+1 and X. What else needs to be adapted depending on the extruder in the example you shared me?

Can we stick to `T0` and `T1` or could we have more ?

Also, for picking last `G1` I am making the assumption in https://github.com/dagoma3d/CuraByDagoma/pull/137/files#diff-1cf8d5b9e4397df1d36cc34515f8f41981ebba572303a84ccb595ccfcb15cff3R67 that we will exactly have a `G1` line on the very previous line of a `;LAYER:` line such as it was in the example. Is it okay?
```
				# G1 F2100 X122.906 Y58.862 E1750.80156
				# ;LAYER:27
```

lissyx, 2023-11-01:
[image]

Just out of print, generated using the WIP in the PR. Just one switch, layer 21 in the UI.

lissyx, 2023-11-01:
After thinking again and given the behavior during this real print, I think there's a bug because we probably have not changed how the finishing step is retracting filament ; my print is finishing with extruder 2 and the filament is still engaged making it impossible for a subsequent print to start if I dont manually remove it. Given the code was not changed, I assume the finishing step is instructing extruder 1 to retract at the end to leave room for the next print.

I am also suspecting https://github.com/dagoma3d/CuraByDagoma/issues/136#issuecomment-1785821886 is hardcoding extruder 1 and extruder 2 but I have not been able to quickly understand what is refering to the extruder there: `G92` ?  `E0` ? This sequence would be OK for switching from extruder 1 to extruder 2, but it is likely wrong if we switch from extruder 2 to extruder 1. Looking at the `Gpreswitch` in the XML I have not been able to figure it out :(

lissyx, 2023-11-02:
(I found a way to disable the button if the user does not have double extruder)

0r31, 2023-11-08:
Hi @lissyx , sorry for the delay.

I hope I can check this week-end (PR and comment).

lissyx, 2023-11-08:
Thanks! FTR I've been printing a lot, and extruder 2 is never retracting its filament at the end of the print :[, at least with my changes. I'm not knowledgeable enough to verify if the GCode is OK, and what it is doing :)

0r31, 2023-11-08:
Could you please describe me a complete scenario?

- The printer choosen in the soft (I guess disco ultimate dual)
- The first extruder to start
- The number of switch during print
- The last extruder to finish

To sum up and make it easier for all, is it possible for you to send me a gcode? (I ask as it can deal with privacy. Theorically, I don't have to know what you are printing).

lissyx, 2023-11-09:
> Could you please describe me a complete scenario?
> 
>     * The printer choosen in the soft (I guess disco ultimate dual)

Yes, Disco Ultimate Dual

> 
>     * The first extruder to start

Extruder #1

> 
>     * The number of switch during print

One switch, around layer 20~ish, to Extruder #2

> 
>     * The last extruder to finish

Extruder #2

> 
> 
> To sum up and make it easier for all, is it possible for you to send me a gcode? (I ask as it can deal with privacy. Theorically, I don't have to know what you are printing).

I'll try to attach here, it's a public model, just a bit big

lissyx, 2023-11-09:
[1xCourbe_R400_30d.zip](https://github.com/dagoma3d/CuraByDagoma/files/13304543/1xCourbe_R400_30d.zip)

0r31, 2023-11-17:
Hi @lissyx ,

I am sorry, not yet :( I will notify you when i could.

lissyx, 2024-01-09:
@0r31 Hello, and happy new year ! I assume lack of reply means you had no time :-).

0r31, 2024-01-19:
Hi, happy new year @lissyx !

I will have, next week ;)

0r31, 2024-01-29:
@lissyx : I am on your PR. Don't see anything weird. The noozle should be freed at the end of the print, regardless the starting extruder and the number of color change.

I am testing your branch with a simple object, starting extruder  = T0 and one color change.
I can see the extruder change :
```
;TYPE:CUSTOM
G92 E0 ;Set current extruder position to 0
G1 E-60 F3000 ;Free the nozzle from filament of extruder T0
T1 ;Switch extruder
G92 E0 ;Set current extruder position to 0
G1 E60 F3000 ;Engage the filement of extruder T1 in the nozzle
G92 E592.66576 ;Set the extruder position to the last known position
```

And after that, no other extruder change occurs, so filament in T1 should be disengaged from the nozzle...

🤔

lissyx, 2024-01-29:
> @lissyx : I am on your PR. Don't see anything weird. The noozle should be freed at the end of the print, regardless the starting extruder and the number of color change.
>
> I am testing your branch [...]

Ok I would need to do more testing, because definitively there's something that does not retract it enough. Maybe there was a regression then ? And I'm just unlucky because applying my patch on top of `main`

0r31, 2024-01-29:
I am doing some real tests on a printer to check.

lissyx, 2024-01-29:
For what it is worth, I'm pretty sure I see a bit of filament retraction at the end, but it looks to be not enough.

0r31, 2024-01-29:
In the gcode end sequence, there is :
`G1 E-60.0 F10000` meaning we retract 6 cm of filament (https://github.com/dagoma3d/CuraByDagoma/blob/a1c6643d2f4fddc0d179083b2d698dea28a82799/resources/xml/discoultimate_dual.xml#L595 and https://github.com/dagoma3d/CuraByDagoma/blob/a1c6643d2f4fddc0d179083b2d698dea28a82799/resources/xml/discoultimate_dual.xml#L668

Could you please send me a gcode with the following params :
- The example object provided by CuraByDagoma
- Start extruder = T0
- One color change

I want to check your end gcode sequence

0r31, 2024-01-29:
@lissyx : I juste merged your PR as i don't see any issue related to the final filament retraction to free the nozzle.

When you have time, could you please send me a generated gcode on your side, as stater in https://github.com/dagoma3d/CuraByDagoma/issues/136#issuecomment-1914318888?

lissyx, 2024-02-02:
Sure, i'll verify this on current stable as well as dev trees

lissyx, 2024-06-25:
(forgot but I'll do it)

0r31, 2025-09-29:
Hi @lissyx ,

I don't work for this company anymore for more than a year so far. You can try to reach any support contact you could find on the whole GitHub account.

Regards,