dagoma3d/CuraByDagoma #139: END code - a ajuster

matt-regnier, 2024-02-06:
changer le END : pour = G1 E-55 F4600 ;retract the filament a bit before lifting the nozzle, to release some of the pressure
[image]

denisdago, 2024-02-07:
@0r31 
J'ai testé 2 fois le gend avec comme consigne de retract : 

G1 E-55 F1200 ;retract the filament a bit before lifting the nozzle, to release some of the pressure

et cela marche parfaitement. donc je préconise ce paramètre (le retract correspond à la poussée de E50 F1200 du gstart de cura by dagoma) 

réflexion : 
les gstart et le gend de cura ultimaker et cura by dagoma sont différent, cela est-il normal ou peut-on avoir les mêmes.
soit on met des gstart et gend identiques pour cura by dagoma et cura ultimaker (si cela est cohérent)
soit on adapte la poussée  de start et le retract de cura by dagoma comme cela si on prépare ses fichiers dans cura by dagoma on peut enchaîner les prints.

0r31, 2024-02-13:
Voir #142